sinara-hw/Humpback #1: [RFC] new EEM: humpback

jordens, 2018-11-18:
Despite the general purpose of the Sinara ecosystem hardware and the wide range of features, lot of development in labs is still based on random "toy"/"development"/"evaluation"/"hobbyist" boards (called "boards" throughout) with µCs and FPGAs. They are dirt cheap, fit for the many applications that lie outside the typical use cases for Sinara EEMs, get the job done just fine, and have a shallow learning curve suitable for physicists. That won't change and there seems no point in trying to replace those "boards". Instead let's try to integrate them.

## Idea

I'd like to propose a simple modification to the [TTL tester EEM](https://github.com/sinara-hw/sinara/raw/master/ARTIQ_ALTIUM/EEMs/Tester/3U_TTL_Tester.PDF) to give it a much wider applicability and open it up for more use cases.

The basic idea is to

1. get connectivity between boards and the Sinara EEM connector
2. build a easy way of mounting those boards in a crate
3. to supply them with power.

## Use cases

* Ethernet connectivity of all kinds. Translate EEM to ethernet.
* Can be get plenty of power from PoE independent of the board (e.g. [21W](https://www.digikey.de/product-detail/en/phihong-usa/POE21-120/993-1080-ND/2384559))
* Wire up microcontrollers that run simple control tasks and loops (temperature control, laser power stabilization, slower modulation/demodulation frequency locks) when (a) where the µC is connected to EEMs and controlling them and (b) when the µC is connected as an EEM to a core device.
* Enable usage and testing of other EEMs with a simple adapter using a µC or FPGA. Also for acceptance testing at the factory.

## Implementations

There seem to be two approaches:

### A EEM adapter: Shield

Build a "shield" that just implements the connectivity aspect (1) where a remote board is connected to the core device.
  * To be used outside a crate
  * 2xRJ45 connected to a DIO-RJ45 EEM in a crate and thus to the core device
  * Should use the same LVDS-to-CMOS converters and DIP switches as Tester. Or maybe just a cheap (ICE40) FPGA (and DIP switches)
  * Should be powered from the board
  * Does not give I2C connectivity.

### A EEM board carrier: Humpback

Build a EEM that achieves (1) both ways, (2), and (3).
  * Call it e.g. "Humpback"
  * Mount the board on the EEM
  * Since there will likely be a need for external connectivity, while the panel pattern for that connectivity is unknown, let's open up the panel and also build a gap into the board so that only a panel frame (or even just a panel-U) remains.
  * Make the EEM short so that small boards can be used and large boards just stick out of the crate.
  * Two EEM connectors
  * Power the board from the EEM
  * Add a barrel connector so that the EEM can power MP3V3, the board and attached EEMs or the EEM and board can be powered from the upstream core device (I.e. dual use just like Tester).
  * Maybe a 5V SMPS for VIN to feed the more [sensitive](https://www.arduino.cc/en/ArduinoCertified/IntelGalileo) boards.
  * Use a cheap FPGA (ICE40 maybe) to convert LVDS. Add DIP switches and implement an I2C direction control shift register in FPGA.
  * Expose 8xCMOS or LVDS raw (for FPGA boards) and I2C to board.
  * Maybe expose MP3V3 to board.

## Potential layouts and pinouts

The choice for the connector layout (power, i2c, digital) is a bit tricky. For both shield and EEM there could be:

### Arduino Uno V3

The "Arduino Uno v3" pattern for the shield and the EEM plus a common denominator of mounting holes and standoffs for the EEM would cover large subset of boards. These are typically stacked and the EEM would probably the "top" of the stack while the board is the "bottom".
Best position and orientation unclear.

µC/CPU:
* [Arduino](https://store.arduino.cc/arduino-due) (*but not with 5V IO!*)
* [ST Nucleo 144](https://www.digikey.com/product-detail/en/stmicroelectronics/NUCLEO-F429ZI/497-16280-ND/5806777), [ST Nucleo 64](https://www.digikey.com/product-detail/en/stmicroelectronics/NUCLEO-F446RE/497-15882-ND/5347712)
* ~[Intel Galileo](https://www.arduino.cc/en/ArduinoCertified/IntelGalileoGen2)~

FPGA:
* Xilinx Artix: [Arty](https://www.xilinx.com/products/boards-and-kits/arty.html#hardware)
* Intel Max 10 (fka Altera) [XLR8](http://www.aloriumtech.com/xlr8-fpga-board/)
* Xilinx Spartan 6: [Papilio (Pro)](http://papilio.cc/index.php?n=Papilio.PapilioPro)
* Lattice ICE40: 

### 26 pin Raspberry Pi+

Has 5V, I2C, 3V CMOS, mounting hole pattern.

* Raspberry Pi B+ etc
* [Orange Pi (Zero)](http://www.orangepi.org/orangepizero/)

### Digilent PMOD

Probably only as a shield.

### ST Morpho

* Nucleo

### ST ZIO

* Nucleo 144

### beagle bone

* Beagle bone black

gkasprow, 2018-11-18:
@jordens do you want one of EEM ports to be upstream? I.e. to use one of SBC as a target for Kasli ?

jordens, 2018-11-19:
3d printed panel inserts sounds good. Then you can easily do recessed panels or none at all for quick testing.
Yes. I'd stick pretty close to the banker design for the eem connectors. Using Humpback as a Target for kasli is definitely a nice use case.

gkasprow, 2018-11-19:
I want to produce Humpback, Stabilizer and Banker at the same time using shared documentation to limit NRE costs.
@jordens @hartytp  please check schematics and initial component placement for Humpback. The board is 99% routed but no DRC and cosmetics were done.
PDF is here, sources are in repo
[Humpback.PDF](https://github.com/sinara-hw/Humpback/files/2596750/Humpback.PDF)

I provided support for: Nucleo 144 (compatible with Arduino), Beaglebone Black, Orange Pi Zero, Wiznet (2x UART only) and ESP32 which uses same code and tools as ESP8266 but has 2 CPU cores and much more memory. I use it for my projects since some time. And it has ufl antenna connector for WiFi and Bluetooth so one can attach SMA pigtail to the 3D printed panel and gain wifi access. It is supported by Arduino so remote logging or control using MQTT with i.e. Cayenne/Blynk and mobile App is trivial - just a few lines of code. I have such projects in my private repo.
All common ports like I2C, UARTs, SPI were routed together to save pins.
All SBCs have access to FPGA configuration pins so they can load the FPGA directly without supporting FLASH. There are open source codes that do it for Beaglebone and STM32.
Beaglebone pin naming comes from LOGI-BONE project where bus access to FPGA is uses. I also use same FPGA config pins so changes to the code would be minimal. Open source HDL and Linux code are available. Rest of Beaglebone pins are named as GPIOs
All dedicated pins were connected to individual FPGA pins. In this way we will be able to provide and support same HDL code for all SBC platforms.

hartytp, 2018-11-19:
@gkasprow I'm not really the best person to review this kind of design. I had a quick look over it and it looks really nice, but I can't give you detailed feedback (not enough experience using these kind of microprocessor boards).

One question: did we plan to make this a PoE module? Or, is it intended to always be powered from a 12V supply?

hartytp, 2018-11-19:
> I want to produce Humpback, Stabilizer and Banker at the same time using shared documentation to limit NRE costs.

Can we sneak Thermostat in there as well?

jordens, 2018-11-19:
A quick first set of comments:

* Great work. Congrats to fitting all devboards onto Humpback! I didn't expect that.
* I assume this would come with all pin headers not soldered but still included.
* It's a bit risky to do all three boards at the same time especially with Banker and Humpback being so similar. Fine if the NRE savings outweigh that risk.
* Testpoints should be on the top if at all possible (TP*, B*)
* Nice solution on the I2C master/slave issue! I also extended the kasli-i2c tools to automatically scan any tree configuration of those I2C switches recursively. I.e. EEPROM detection is fine at any depth.
* The Nucleo 144 boards seem to have one of four different pinouts. But I couldn't identify the actual difference quickly. Which pinout is this?
* Is that distance between the capacitor/resistor combo and the front panel screws/washers large enough?
* Doesn't the ESP32 need a specific ground/keep-out area/pattern near the PCB antenna?
* From the PCB renders with the 12V copper areas, the EEM headers are still the wrong assignment. The upstream one should be the rightmost.
* Is 10µF as the biggest cap going to be sufficient for the FPGA VCC/VCCIO? Comparing that with Kasli it sounds tiny. But I guess you took a devboard as the reference. 
* I didn't check any pinouts, or pins vs datasheet consistency. Would you like someone to do that?
* No PoE is fine by me. There are cheap external PoE+ splitters for barrel.

gkasprow, 2018-11-19:
> 
>     Great work. Congrats to fitting all devboards onto Humpback! I didn't expect that.
>     I assume this would come with all pin headers not soldered but still included.

yes

>     It's a bit risky to do all three boards at the same time especially with Banker and Humpback being so similar. Fine if the NRE savings outweigh that risk.

these are relatively simple boards, easy to rework

>     Testpoints should be on the top if at all possible (TP*, B*)

Aren't they? I'll check

>     The Nucleo 144 boards seem to have one of four different pinouts. But I couldn't identify the actual difference quickly. Which pinout is this?

I took the 144 series because they have embedded Ethernet

>     Is that distance between the capacitor/resistor combo and the front panel screws/washers large enough?

I'll check

>     Doesn't the ESP32 need a specific ground/keep-out area/pattern near the PCB antenna?

I used one with ufl. An antenna inside of the rack do not make a lot of sense.

>     From the PCB renders with the 12V copper areas, the EEM headers are still the wrong assignment. The upstream one should be the rightmost.

that's right, the copper is not finished, it's copied from previous version of Banker

>     Is 10µF as the biggest cap going to be sufficient for the FPGA VCC/VCCIO? Comparing that with Kasli it sounds tiny. But I guess you took a devboard as the reference.

This FPGA consumes really tiny amount of power, I'll check with datasneet. I copied it from another design.

>     I didn't check any pinouts, or pins vs datasheet consistency. Would you like someone to do that?

Yes, mainly power supplies and dedicated pins. Rest of the are general purpose IOs which can go to any location

>     No PoE is fine by me. There are cheap external PoE+ splitters for barrel.

right
>

gkasprow, 2018-11-19:
I took 144 series devkit with CPU we use for Stabilizer as a reference.

jordens, 2018-11-20:
I'll try to go through the power supplies and dedicated pins over the next days. Ack the other points.

jordens, 2018-11-20:
* I'd suggest connecting GBIN0 to SPI_SCK (clock FPGA from CPU SPI) and GBIN1 to STM32_PB10 or STM32_PB11 (clock FPGA from CPU timer)
* On the beaglebone I think you have CFG_MOSI and CFG_MISO swapped. On STM32 MISO/MOSI can be swapped. I don't know about BB.
* If the I2C tree gets large, multiple 2k2 resistors are in parallel (up to eight counting Humpback/Banker alone if the switch has all ports enabled, more if there are EEMs connected to it). Is this really robust? How many I2C bus segments can you enable until you get a latchup? I think that note on the schematic only applies to banker where the FPGA bank is at 2.5V.
* GBIN7 must be the first LVDS pair of the upstream EEM (both here and on banker)
* GBIN6 can be the first downstream EEM

gkasprow, 2018-11-21:
> 
>     I'd suggest connecting GBIN0 to SPI_SCK (clock FPGA from CPU SPI) and GBIN1 to STM32_PB10 or STM32_PB11 (clock FPGA from CPU timer), GBIN4 and 5 are good.

OK

>     On the beaglebone I think you have CFG_MOSI and CFG_MISO swapped. On STM32 MISO/MOSI can be swapped. I don't know about BB.

will check it

> If the I2C tree gets large, multiple 2k2 resistors are in parallel (up to eight counting Humpback/Banker alone if the switch has all ports enabled, more if there are EEMs connected to it). Is this [...]

True, actually I don't think that somebody will enable all ports, but when one does, may lock the bus.

>     GBIN7 must be the first LVDS pair of the upstream EEM (both here and on banker)

OK 

>     GBIN6 can be the first downstream EEM
>     Probably worth running the Lattice tools to estimate dynamic power consumption.

The design must contain some useful logic to make sense doing it.

>     The 1.2V LDO talks about a 2 mA minimum current draw. The static draw from the FPGA is 1.1 mA.

Good catch, anyway in real life it works and voltage is 1.2V. I can add some load to be sure.

>     There is a GND pin on the ESP32 not connected.

ok

>     Is T8A/T8B just two AND gates with open collector outputs? What's the story behind it?

simplicity. It is used on all ESP32 devkits so I don't want to invent something else.

>     What's with SW0 and SW1? And separate from that, why are FLASH_SDI/SCK on the dip switch? How would you use that?

After FPGA gets configured, these are regular IOs and can be used as inputs. In this way you spare 2 pins.

>     Does it make sense to put populated 0R resistors into at least a few of the ADC and DAC pins of the STM32 to optionally disconnect them from the FPGA (FPGA pulling the ADC inputs, or coupling noise)? Similar for BB.

Do we plan to use thes ADC and DAC pins for something analog?

>     There seems to be an opportunity to remove a lot of wire crossings by pin swapping, e.g. around the DIP switches. I guess that's what you meant by cosmetics.
>
Yes, but vias cost nothing. I meant curly and dangling traces :)

jordens, 2018-11-21:
All ACK.
I'd like to be able to use the STM32 ADC/DAC without worrying about the FPGA pins connected to them. Let's put 0R (or a DIP switch array? but I am fine with 0R here) on PA3 PC0 PC3 PC1 PC4 PC5 and PB1 PC2 PA2 for ADC and PA4 PA5 for DAC.

jordens, 2018-11-21:
And it would be great if the various switch and pin arrays could get lots of silkscreen labels, even for individual pins on the PCB top side.

jordens, 2018-11-22:
Looking at the [ice40hx break out board](https://www.latticesemi.com/view_document?document_id=50373) and the [ice40 hardware checklist](http://www.latticesemi.com/-/media/LatticeSemi/Documents/ApplicationNotes/IK/iCE40HardwareChecklist.ashx?document_id=47779). Probably also applies to Banker.
* The FPGA gets a 100n for each supply pin
* 10u for VCCPLL1 and 0
* You decided that LDOs to get from 5V to 1.2 and 2.5V is efficient enough and not switchers are needed?
* They have 10n and 1u on a couple of rails but I don't know whether that makes much sense. THey don't have your 100p though.
* N14 on the FPGA is TRST_B on that devboard and pulled low. Also mentioned explicitly in the HW checklist.

gkasprow, 2018-11-22:
I copied it from working design. And that working design was copied from somewhere else. So errors could propagate.
I can add switcher for 1.2V, but for such small FPGAs I expect core current less than 0.5A.
We can use same low cost AUR9705 chip as for 3.3V

jordens, 2018-11-22:
ACK, I am not particularly worried about LDO vs switcher and my guess is significantly less than 0.5A but that would be 1W burned in the LDO.

gkasprow, 2018-11-23:
I didn't finish the mechanical side yet. True, we will need Ethernet cutout to use stadard goldpin connectors.

gkasprow, 2018-11-23:
Lattice is using 300mA LDO (LT3030EFE#TRPBF) on their ICESTICK devkit which uses  HX4K FPGA They supply it from 5V directly.

gkasprow, 2018-11-25:
OK, I used switcher. It costs 0.5EUR.

jordens, 2018-11-27:
@gkasprow Thanks for the new RC. A few things left from above. Ignore if you are already aware.
> * I'd suggest connecting GBIN0 to SPI_SCK (clock FPGA from CPU SPI) and GBIN1 to STM32_PB10 or STM32_PB11 (clock FPGA from CPU timer), GBIN4 and 5 are good.
> * Beefing up/bypassing of the downstream EEM fuses
> * RJ45-PCB interference and beaglebone mounting?

* On Humpack the front panel screw RC combo is now nicely clear of the screw/washer. Should that also be implemented for Banker?

gkasprow, 2018-11-27:
@jordens there is no place for RJ45 cutout. It collides with other board connector. Let's use longer headers.
Bypassing EEM fuses can be good idea. It will simplify layout and improve SI of LVDS lines.

jordens, 2018-11-27:
Yes. Let's skip the eem fuses on Humpback Altogether. Ack the longer headers for beahlebone. does it also apply to orange pi zero?

gkasprow, 2018-11-27:
Orange pi has no connector soldered by default so standard pins would work.

jordens, 2018-12-02:
The stackup has a 0,71mil dielectric between Top and Gnd layers. I hope that's a typo.