sinara-hw/Humpback #3: panel

gkasprow, 2019-01-09:
@jordens Please have a look at the panel
[image]

jordens, 2019-01-10:
Ok. Nice.
* I guess the screw holes are for mounting the panel insert, right?
* What about removing the panel "bar" at the top (right side when in the crate)? Then a front panel could extend beyond 8 HP and would already be mechanically fixed to Humpback.

gkasprow, 2019-01-10:
Yes, they are for fixing the panel insert. I can remove the panel bar, but I'm afraid it won't be rigid enough and bend. Why do you want to extend beyond 8HP?

jordens, 2019-01-10:
To expose e.g. the µC ADC and DAC channels on the front panel. But you are probably right, that's not particular rigid, the use cases for raw ADC and DAC channels are few and if needed one coule hack up a quick 3U raster board, some jumper cables and connectors with its own panel.

closed 2019-01-10 by gkasprow

jordens, 2019-01-11:
@gkasprow I think the panel might be covering the edge LEDs on the underside of the board. Add holes?