sinara-hw/Humpback #6: Humpback tests

gkasprow, 2019-03-05:
Humpback arrived!
[image]

gkasprow, 2019-03-06:
power supply works fine after R166 fix

gkasprow, 2019-03-07:
LED blinks, FLASH works
I used [same FPGA](https://github.com/sinara-hw/Banker/issues/8) project as for Banker

jordens, 2019-03-13:
Also works fine mechanically and electrically with a Nucleo144 STM32F429 and ethernet.

jordens, 2019-03-13:
* I'd add the headers for the boards to the BOM but not mounted, just shipped.
* Make the panel cutout 0.5 mm wider. The nucleo barely fits.
* I would not mount the ESP32 by default. It's a bit tricky to remove.
* Also on the ESP32, I think that's the version with the wifi signal connected to the u.fl connector and not to the PCB antenna by default. IMO that's ok but **if** there is a ESP32 mounted, there should be a pigtail antenna.
* JP2 and W1 should get silkscreen labels.