sinara-hw/Humpback #7: Humpback errata

gkasprow, 2019-03-06:
Pull R166 to P12V0 instead of P3V3

kaolpr, 2020-05-21:
@gkasprow What do you mean by "add R to GND"?

gkasprow, 2020-05-21:
@kaolpr just make voltage divider between 12V and GND that gives 3.3V.
For the moment, to fix v1.0 remove R160 and connect pin 38 of IC87 via 10k to +12V
[image]

In the next revision make sure IC2 and IC3 are supplied from 12V via parallel Zener-based regulator.

gkasprow, 2020-05-21:
For rev 1.0 just cut the polygon and connect the LDO input to 5V rail
[image]

gkasprow, 2020-05-21:
Optionally, one can lift pin 3 of IC4 and connect it to C244 without cutting the polygon.

jordens, 2020-05-21:
@kaolpr  is this for a new batch of v1.0 or a new revision?

gkasprow, 2020-05-21:
TS produced more v1 boards and need to implement these fixes manually