sinara-hw/Humpback #11: Bottom part of STM32 nucleo blocks downstream0 port

occheung, 2020-07-02:
[image]
[image]
[image]

gkasprow, 2020-07-03:
Just solder correct stacking height connectors! This boards was not supposed to be plugged directy.

sbourdeauducq, 2020-08-12:
Do you have part numbers for suitable connectors? With "regular" ~1cm ones, the Nucleo board can be mounted but the EEM connector is obstructed.
BTW this is less of a problem for Nucleo h743zi (as opposed to h743zi2) since the JTAG module can be detached on h743zi.

gkasprow, 2020-08-12:
I'm using elevated connectors for boards plugged to RPI: BSSH-113-D-10-G or ESW-113-23-S-D

gkasprow, 2020-08-12:
they mate with standard female 2.54 sockets

sbourdeauducq, 2020-08-13:
Ordering:
* 16 pins: Mouser No:200-ESW10823GD; Mfr. No: ESW-108-23-G-D
* 20 pins: Mouser No:200-ESW11023GD; Mfr. No: ESW-110-23-G-D
* 30 pins: Mouser No:200-ESW11523GD; Mfr. No: ESW-115-23-G-D
* 34 pins: Mouser No:200-ESW11723GD; Mfr. No: ESW-117-23-G-D

sbourdeauducq, 2020-09-01:
OK, this works. Please close

closed 2020-09-01 by occheung

gkasprow, 2020-09-01:
I will add the info to the schematics